OSU-CMS/Ph2_ACF_GUI #95: Create config generator

ryansantos1174, 2023-09-21:
When other institutions clone the repository we do not want them to have to recreate their config file every time they do a pull request. We also want the config file to be updated with any new variables that we decide that we need. Therefore, we cannot just add this to the list of gitignore files. We should create a formatted text file (csv, xml, toml, etc. ) that stores any site configuration, then create a python function that will generate the necessary python configuration file based off of the text file. This function should then be called inside QtApplication before the GUI is ever launched.

ryansantos1174, 2023-09-27:
Looking at the current state of the branch, it looks like you used the wrong settings file. Instead of siteSettings.py , you based it off of GUIutils/settings.py. Also the way that you have your code setup, you would expect a csv file of the form 

variable1 variable2 variable3 ..
value1 , value2, value3 ...

so you would have  a ton of columns
I was thinking more something like:
variable, value
variable1, value1
variable2, value2

davidkim918, 2023-10-13:
I reopened the issue because I added docstrings and comments, a few edits to the code, and moved the file so settingimport.py will be within the same directory as the siteSettings_template.py and siteSettings.py