OSU-CMS/Ph2_ACF_GUI #120: memory leak in main branch

CollinKa, 2023-10-23:
I noticed an issue on the main branch that may related to a memory leak. When I keep changing the serial number multiple times, the GUI becomes laggy. The memory and CPU usage on GUI reaches 100%. This may have to do whenever we type the serial number, GUI creates a new start window. DEV branch doesn't have this issue.

mlj5j, 2023-10-26:
Thanks for finding this and documenting it this way.  I think this is a previously unknown symptom of a known issue that was fixed in the DEV branch (There was a widget that wasn't being deleted properly).  We'll be doing a merge of DEV -> main soon, so we can check to see if it is solved after that.

ryansantos1174, 2023-12-01:
Fixed when main was updated with DEV branch

closed 2023-12-01 by ryansantos1174