OSU-CMS/Ph2_ACF_GUI #310: Manual power supply control incorrectly seen as activated

clelange, 2024-09-25:
I'm trying to run a calibration as localexpert. After connecting the FC7 and clicking on "New", I get the error shown in the screen shot. Then the "New" button is disabled and I cannot really do anything useful anymore.

[image]

Log output:

```shell
[cmspixel@PC16525]~/Ph2_ACF_GUI% bash run_Docker.sh
ls: cannot access '/dev/ttyUSB*': No such file or directory
device loaded /dev/bus/usb/
device loaded /dev/ttyACM1
device loaded /dev/ttyACM2
device loaded /dev/ttyACM0
--device=/dev/bus/usb/ --device=/dev/ttyACM1 --device=/dev/ttyACM2 --device=/dev/ttyACM0
running as user
latest: Pulling from majoyce2/ph2_acf_gui_user
Digest: sha256:0d7b424458088edb845a3955492d3548bdd688321a9f391604696ec106205d9a
Status: Image is up to date for majoyce2/ph2_acf_gui_user:latest
=== DONE: you can now run cmake ===
You can now open the GUI by doing 'python3 QtApplication.py'.
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
/usr/local/lib/python3.9/site-packages/pyvisa_py/tcpip.py:122: UserWarning: TCPIP::hislip resource discovery requires the zeroconf package to be installed... try 'pip install zeroconf'
  warnings.warn(
2024-09-25 13:26:35,465 - Gui.python.logging_config - WARNING - Initialized variables for QtApplication
Initialized variables for QtApplication
MESA-LOADER: failed to open swrast: /usr/lib64/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib64/dri, suffix _dri)
failed to load driver: swrast
('ASRL/dev/ttyACM0::INSTR', 'ASRL/dev/ttyACM2::INSTR', 'ASRL/dev/ttyACM1::INSTR')
FwModule is: ['fc7-1']
```

`siteConfig.py` content:

```python
import json
import logging

# Customize the logging configuration
logging.basicConfig(
   level=logging.INFO,
   format='%(asctime)s - %(name)s - %(levelname)s - %(message)s',
   filename='my_project.log',  # Specify a log file
   filemode='w'  # 'w' for write, 'a' for append
)

logger = logging.getLogger(__name__)

######################################################################
# To be edited by expert as default setting for Hardware configuration
######################################################################

# Set gpib debug mode to True if you want to see SCPI commands sent to HV and LV devices:
GPIB_DebugMode = False

##  The following block sets defaults that are to be used in the simplified (non-expert) mode.  ##

# default mode for LV powering (Direct,SLDO,etc)
defaultPowerMode = "SLDO"

##################################

#default BaudRate for Arduino sensor
defaultSensorBaudRate = 9600
defaultArduino = "Arduino SA Uno R3 (CDC ACM) ACM0"

#UIC coldbox variables
usePeltier = False
defaultPeltierPort = '/dev/ttyUSBPeltier'
defaultPeltierBaud = 9600
defaultPeltierSetTemp = 10
defaultPeltierWarningTemp = 40
#################################

# Icicle variables

# Set this variable to use your powersupplies manually
# IF THIS VARIABLE IS SET, THEN ICICLE_INSTRUMENT_SETUP WILL NOT BE USED
manual_powersupply_control = False

# Load instrument setup from json file
with open('jsonFiles/instruments_psi_pc16525.json', 'r') as file:
    icicle_instrument_setup = json.load(file)

#Set peak voltage for bias scan.  Make sure this value is negative or it could damage the sensor.
IVcurve_range = -80 #Maximum voltage in Volts to be used in IVcurve

forward_bias_voltage = 0.5 #positive voltage used to run a forward-reverse bias bump bond test

## Update this dictionary for the IP addreses of your FC7 devices ##
FC7List =  {
	'fc7-1'	:	'192.168.1.80',
	'fc7-2'	:	'192.168.1.81',
}

## Update this dictionary for your simple mode cable mappings ##
CableMapping = {
    "0" : {"FC7": "fc7-1", "FMCID": "L12", "FMCPort": "0"},
    "1" : {"FC7": "fc7-2", "FMCID": "L12", "FMCPort": "0"}
}

## Specify whether of not you want to monitor chip temperature during the tests ##
## Set this to "1" if you want the monitoring enabled.  Set it to "0" if you want it disabled. ##
Monitor_RD53A = "1"
Monitor_CROC = "0"
Monitor_SleepTime = "30000"  # time in milliseconds between temperature readings

## Establish thresholds for chip temperature readings. A chip reading above the
## emergency threshold will abort the test.
Warning_Threshold = 25
Emergency_Threshold = 40

## Configuring the current settings for each module type.  These values are in Amps. 
ModuleCurrentMap = {
	"SCC" : 0.6,
	"TFPX RD53A Quad" : 6.5,
	"TEPX RD53A Quad" : 6,
	"TBPX RD53A Quad" : 6.5,
	"TFPX CROC 1x2"  : 4.5,
	"TFPX CROC Quad" : 7.5,
	"CROC SCC"  : 2.0,
	"TEPX CROC 1x2"  : 4.5,
	"TEPX CROC Quad" : 7.5,
	"TBPX CROC 1x2"  : 4.5,
	"TBPX CROC Quad" : 7.5,
}

## Configuring the voltage limit for each module type when operating in SLDO mode.  These values are in Volts.
ModuleVoltageMapSLDO = {
	"SCC" : 1.8,
	"TFPX RD53A Quad" : 2.98,
	"TEPX RD53A Quad" : 2.0,
	"TBPX RD53A Quad" : 2.98,
	"TFPX CROC 1x2"  : 2.3,
	"TEPX CROC 1x2"  : 2.3,
	"TBPX CROC 1x2"  : 2.3,
	"TFPX CROC Quad" : 2.98,
	"TEPX CROC Quad" : 2.98,
	"TBPX CROC Quad" : 2.98,
	"CROC SCC"  : 1.8,
}

###### Should not be using direct voltage, so this block can probably be removed #####
##  Configuring the voltage settings for each module type.  These values are in Volts.
ModuleVoltageMap = {
	"SCC" : 1.3,
	"CROC SCC"  : 1.6,
}
#####################################################

######  can probably remove this block  #############
#setting the sequence of threshold tuning targets:
#defaultTargetThr = ['2000','1500','1200','1000','800']

##### The following settings are for SLDO scans developed for Purdue.#####
##### Do not modify these settings unless you know what you are doing.####
#default settings for SLDO scan.
defaultSLDOscanVoltage = 0.0
defaultSLDOscanMaxCurrent = 0.0
#####################################################
```

`jsonFiles/instruments_psi_pc16525.json` content:

```json
{
    "instrument_dict": {
        "lv_1": {
            "class": "HMP4040",
            "resource": "ASRL/dev/cldbxlv1::INSTR",
            "sim": false,
            "default_voltage": 2.5,
            "default_current": 7.2
        },
        "lv_2": {
            "class": "HMP4040",  
            "resource": "ASRL/dev/cldbxlv2::INSTR",
            "sim": false,
            "default_voltage": 2.5,
            "default_current": 7.2
        },
        "hv": {
            "class": "CaenDT8033N",
            "resource": "ASRL/dev/dt803x::INSTR",
            "sim": false,
            "default_voltage": 80,
            "default_current": 5e-6
        },
        "chiller": {
            "class": "Lauda",
            "resource": "ASRL/dev/ttyUSB0::INSTR",
            "sim": false,
            "default_voltage": 80,
            "default_current": 5e-6
        }
    },
    "channels_dict":{
        "0":{
            "lv": {"instrument":"lv_1", "channel":1},
            "hv": {"instrument":"hv", "channel":1}
        },
        "1":{
            "lv": {"instrument":"lv_2", "channel":1},
            "hv": {"instrument":"hv", "channel":1}
        }
    }
}
```

clelange, 2024-09-25:
I then refreshed and a new window showed up. I entered a serial number but then get the following issue:

[image]

mlj5j, 2024-10-01:
For the first problem, did you click "Connect all devices"?  I think that might resolve the problem.  But it should have opened the next window anyway (maybe it's behind the current window so it's hidden?).

Second problem, I see what's happening here.  The "P" prefix isn't listed as a known module type so it complains about it, but it should still run.  I need to map it to a module type and that should fix it.  Is this prefex the same for all TEPX CROC QUAD modules?  But the reason it won't run is because you need to input which FMC port number you are using.  It will complain about the module not being found in the database but if you click okay it should still let you run it.

[image]

Then I click "OK" and it goes to here
[image]

mlj5j, 2024-10-01:
I see two other things I need to fix.  The addition of the CAEN and the chiller needs some attention.  We'll start working on these.

clelange, 2024-10-09:
Thanks for the feedback.

We decided to use "P" as prefix for modules used at PSI. Those built in other places will have different prefixes (I need to look them up) to be sure.
When using the CAEN, ICICLE doesn't seem to know about it. Does this require a new docker image to be built?
We can live without the chiller for now. It'd be OK to even only run with LV and no HV for now. Not sure I can actually not have an HV device in the config. Any advice?

mlj5j, 2024-10-09:
Yes, I need to build the new image this afternoon.  That will have the CAEN in it but not the chiller.  You will get a complain that it can't find your module in the database (for now it tries to pull the trim values from the US database) but you can enter them manually and it should work.

closed 2025-01-30 by mlj5j