OSU-CMS/Ph2_ACF_GUI #438: LV Power Supply sets same current and voltage no matter the number of modules connected in serial powering

lzygala, 2025-05-20:
We are planning to power 8 modules in series with a single LV power supply. When testing with just 2 modules, we noticed that the voltage and current set by the LV power supply is the same no matter how many modules are included in the test (we have been testing using PixelAlive).

lzygala, 2025-06-02:
@mlj5j I have done some measurements of the voltage drop across the modules we have here at Cornell (we have three 1x2s and one 2x2). 

In the attached document you will see the measurements for configurations of 2, 3, and 4 modules connected in series. The cable position numbers in the table translate directly to the ports on our serial cable in which the modules are connected. The largest drop that I measured when we had all 4 modules connected was 1.47V between the first and fourth module.

Please let me know if you need any more information

[Cornell TFPX module testing electronics - Voltage Drop.pdf](https://github.com/user-attachments/files/20558272/Cornell.TFPX.module.testing.electronics.-.Voltage.Drop.pdf)

ryansantos1174, 2025-07-07:
The voltage limit can be changed in siteConfig.py by user to accommodate multiple modules.