OSU-CMS/Ph2_ACF_GUI #499: [BUG] Failed to upload to Panthera

webberfrog, 2025-06-10:
**Describe the bug**
After TFPX_FullPerformance_Test on RH0106, results did not upload to panthera. Got the error message 'testHandler.collect_plots Exception: FileNotFoundError(2, 'No such file or directory')' on the terminal output.

Bug occurred on UIC2 testing station Using v2.2.2 tag

mlj5j, 2025-06-11:
So the failure was just when you clicked "Upload"?  Is that right?

mlj5j, 2025-06-17:
This issue was fixed through a felis update.  The problem was caused by the dictionary in the upload json file created by felis being appended instead of reinitialized when running consecutive sequences.  This caused repeating indicies which caused the upload failure.

closed 2025-06-17 by mlj5j