OSU-CMS/Ph2_ACF_GUI #604: GUI crashes when trying to start a new test

PeterNKim, 2025-08-20:
We are trying to run a pixelalive scan on a TFPX quad but the gui crashes when trying to start a new test. I've attached the json file and the siteConfig.py file we use for this as well. The error we get right after the gui crashes is:

>Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib64/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib64/python3.9/site-packages/ROOT/_application.py", line 91, in _process_root_events
    gSystem.ProcessEvents()
ReferenceError: attempt to access a null-pointer

We are only using an LV source for this. And the only things we've touched are the 2 files listed below. Thanks in advance.

[siteConfig.py](https://github.com/user-attachments/files/21906930/siteConfig.py)

[instruments_osu_oneLV.json](https://github.com/user-attachments/files/21906944/instruments_osu_oneLV.json)

mlj5j, 2025-08-21:
@PeterNKim I see the issue.  The GUI uses the Icicle instrument control package to communicate with the power supplies.  Icicle has an interlock built in that requires both HV and LV devices.  It checks the status of both before issues commands.  Specifically before turning off the LV it requires that the HV be turned off.  We supported manual power supply control in the past, but once we thought everyone had the HV and LV hardware in place we stopped testing operation in manual mode.  If you have an HV device that isn't currently supported we could work on that.  Otherwise I can see if we can bring back the manual control support and you'll have to manually control your LV device.  There might be a way to bypass the interlock when an HV device isn't being used, but I'll have to look at this more closely.

PeterNKim, 2025-08-21:
@mlj5j Ah I see. We do have a supported HV device (I think) but since we're not using it for our scans at the moment, but should we just add it back into the json file?

PeterNKim, 2025-08-21:
We have a Keithley 2400 for our HV, and a TTi QL355TP for our LV. I don't seem to see these (for the specific Keithley at least) in the instrument cluster. Could that be an issue as well when we try to connect our devices?

PeterNKim, 2025-08-21:
I changed the json file (attached at bottom) to now where it has an hv source, and switched the classes of the devices. It seems to have done something since I didn't get immediately get an error saying to check device connections, but it still did time out. Here's the traceback:

>2025-08-21 20:59:12,592 - Gui.python.logging_config - ERROR - Traceback (most recent call last):
  File "/home/cmsTkUser/Ph2_ACF_GUI/Gui/QtGUIutils/QtApplication.py", line 984, in connect_devices
    self.instruments.open()
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument_cluster.py", line 1336, in open
    self.__enter__(*args, **kwargs)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument.py", line 257, in wrapper
    rval = function(self, *args, **kwargs)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument_cluster.py", line 440, in __enter__
    self.hv_set_ocp_module(module, module["hv"].default_current, no_lock=True)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument.py", line 257, in wrapper
    rval = function(self, *args, **kwargs)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument_cluster.py", line 1037, in hv_set_ocp_module
    module["hv"].current_limit = compliance_current
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/keithley2410.py", line 431, in current_limit
    self._instrument.set("SENSE_CURRENT_RANGE", value)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument.py", line 257, in wrapper
    rval = function(self, *args, **kwargs)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument.py", line 321, in wrapper
    return function(self, *args, **kwargs)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/scpi_instrument.py", line 183, in set
    ret = self.query(setting, *value, no_lock=True, attempts=1)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument.py", line 257, in wrapper
    rval = function(self, *args, **kwargs)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/instrument.py", line 310, in wrapper
    return function(self, *args, **kwargs)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/scpi_instrument.py", line 255, in query
    ret = helper.read(self._instrument)
  File "/home/cmsTkUser/Ph2_ACF_GUI/icicle/icicle/utils/setting_processors.py", line 41, in _read
    return instrument.read().strip()
  File "/usr/local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 485, in read
    message = self._read_raw().decode(enco)
  File "/usr/local/lib/python3.9/site-packages/pyvisa/resources/messagebased.py", line 441, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python3.9/site-packages/pyvisa_py/highlevel.py", line 520, in read
    return data, self.handle_return_value(session, status_code)
  File "/usr/local/lib/python3.9/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

[instruments_osu_oneLV.json](https://github.com/user-attachments/files/21926246/instruments_osu_oneLV.json)

ryansantos1174, 2025-09-03:
Just double checking, you have a Keithley 2400 and not a 2410 or 2470 right? As you may have already noticed, the only devices supported by instrument cluster in icicle are given here [https://gitlab.cern.ch/icicle/icicle/-/blob/master/icicle/instrument_cluster.py?ref_type=heads#L103](url) . Usually you get the error that you are getting when you were able to reach a device but the communication process didn't occur properly.

 Best case scenario is that you just have the wrong port or the wrong baud rate and termination setup. Double check that you have put in the correct port. The baud rate should also be set to 9600 on the Keithley and the read and write termination should be set to "\n" (ie. LF). 

If this still doesn't work, then it may be that the Keithley 2400 is just different enough from the Keithley 2410 that it requires its own implementation in icicle.